devbisme/KiField #1: UnicodeDecodeError with default ascii encoding

thomashenauer, 2016-02-26:
I have a "μ" in my part description and this caused KiField to fail with following error: 

> UnicodeDecodeError: 'ascii' codec can't decode byte 0xce in position 14: ordinal not in range(128)

I fixed this by adding following lines in kifield.py:

> reload(sys)
> sys.setdefaultencoding('utf8')

I understand the setdefaultencoding() usage is not encouraged and the reload of sys is a workaround. But in the end it worked for me. See following link: http://stackoverflow.com/questions/3828723/why-should-we-not-use-sys-setdefaultencodingutf-8-in-a-py-script

I use python 2.7 and KiField 0.1.1 on OS X 10.11.3

DefProc, 2016-09-14:
I hit this problem too from using `µ` symbols in the schematic. While it looks like the default format for the schematic files is utf8, I at least got round the problem by running: 

`grep --color='auto' -P -n "[\x80-\xFF]" my_schematic.sch` 

([source](http://stackoverflow.com/questions/3001177/how-do-i-grep-for-all-non-ascii-characters-in-unix#9395552))

And replacing all the `µ` symbols with `u`

kasbah, 2016-09-14:
Seems like you could also use Python 3 to avoid this error for the time being.

PR #16 (2016-10-24): Set default encoding to utf8
This fixes #1. I played around a while trying to set `encoding='utf-8'` in all the csv `reader` and `writer` constructors and `open` calls and using the unicodecsv module. Nothing seemed to work.  

I understand this is essentially a hack but it does work. I believe it's a no-op in Python 3.

https://stackoverflow.com/questions/3828723/why-should-we-not-use-sys-setdefaultencodingutf-8-in-a-py-script
commit d8edeaa9 (2016-10-24): Set default encoding to utf8

This fixes #1. I played around a while trying to set `encoding='utf-8'`
in all the csv `reader` and `writer` constructors and `open` calls and
using the unicodecsv module. Nothing seemed to work.

I understand this is essentially a hack but it does work. I believe it's
a no-op in Python 3.

https://stackoverflow.com/questions/3828723/why-should-we-not-use-sys-setdefaultencodingutf-8-in-a-py-script
closed 2016-10-24 by xesscorp