devbisme/KiField #10: TypeError: "delimiter" must be string, not unicode

cioma, 2016-09-25:
Latest KiField version gives the following error when inserting from CSV to SCH:
TypeError: "delimiter" must be string, not unicode

It seems to be related to this: [http://stackoverflow.com/questions/5625412/python-csv-module-error](url)

kasbah, 2016-09-25:
Are you saying you didn't get this with an older version?

I did some work on support tab delimters. The odd thing is is we get the delimter from the csv parse module itself. 

https://github.com/xesscorp/KiField/blob/2950be7875ec555fded0fe3a8c109565b0096dff/kifield/kifield.py#L125

cioma, 2016-09-25:
I think it worked OK in v 0.1.1. My test file is CSV with comma as separator. It seems there is some problem in the Python's csv module sniffer itself.

kasbah, 2016-09-25:
Could you try and downgrade to 0.1.3? 

```
pip install kifield==0.1.3
```

Another thing to try would be to switch to Python 3.

cioma, 2016-09-26:
v0.1.3 works OK so it's only v0.1.4 that gives aforementioned error while importing from CSV to schematic

PTaylor-RA, 2016-10-14:
I'm using 0.1.4 and for several conversions back and forth it worked fine, then all of a sudden it started giving this error.

I'm using libreOffice calc and have tried save as with both UTF-8 and Western Europe (ASCII/US) character sets.

PTaylor-RA, 2016-10-15:
I saved the csv as an xlsx and it appears to be working.  It looks like the error occurs while converting the csv to wb.

PR #15 (2016-10-24): Work around problem with csv and unicode_literals
This fixes #10
commit 41528475 (2016-10-24): Work around problem with csv and unicode_literals

fixes #10
closed 2016-10-24 by xesscorp