devbisme/KiField #17: Python -- Could not determine delimeter

4b11b4, 2016-10-23:
Hello and thank you for the work on KiField -- you have saved many hours of wasted time for people all over the world.

I'm unable to use KiField for some reason with .tsv's and .csv's, but a .xlsx file works.
Python explains that the csv.Sniffer() raises the error "Could not determine delimeter" (the output message is below).

Is there any other debugging I can do? The --debug flag doesn't produce any more information.

Attached are the .sch, .tsv, .csv, and .xlsx files in a .zip.
[python-delimeter_repro-files.zip](https://github.com/xesscorp/KiField/files/546109/python-delimeter_repro-files.zip)

Mint 18 16.04
KiField 0.1.4

`ab@ab-mint1604 ~/src/folktek/official/33-34_synesthesia/34_comp $ kifield --extract fields.csv --insert envelope-gen_LM358_DFN8.sch --overwrite --debug
Traceback (most recent call last):
  File "/usr/local/bin/kifield", line 9, in <module>
    load_entry_point('kifield==0.1.4', 'console_scripts', 'kifield')()
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.4-py2.7.egg/kifield/__main__.py", line 147, in main
    exc_field_names=exc_fields)
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.4-py2.7.egg/kifield/kifield.py", line 998, in kifield
    part_fields_dict = extract_part_fields(extract_filenames, inc_field_names, exc_field_names)
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.4-py2.7.egg/kifield/kifield.py", line 574, in extract_part_fields
    f_part_fields_dict = extraction_functions[f_extension](f, inc_field_names, exc_field_names)
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.4-py2.7.egg/kifield/kifield.py", line 321, in extract_part_fields_from_csv
    wb,_ = csvfile_to_wb(filename)
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.4-py2.7.egg/kifield/kifield.py", line 125, in csvfile_to_wb
    dialect = csv.Sniffer().sniff(csv_file.read(1024))
  File "/usr/lib/python2.7/csv.py", line 188, in sniff
    raise Error, "Could not determine delimiter"
_csv.Error: Could not determine delimiter
`
[python-delimeter_repro-files.zip](https://github.com/xesscorp/KiField/files/546112/python-delimeter_repro-files.zip)

kasbah, 2016-10-26:
Hey @4b11b4, I think I have fixed your issue with PR #18. 

For future reference I believe you can get the most debugging info by setting `--debug=3`. Mostly I just use `print` where I need to and remove it before committing. The way I get a quick edit test cycle is by doing something like:

```
pip install -U <path to KiField> && kifield --extract foo.sch -i bar.csv
```

Which installs the latest (edited) code and runs my test in one swoop.

libbil, 2016-11-25:
Hi, I might be a bit late to the party but I also had the same error for some csv files. I searched around and found a solution [here](http://stackoverflow.com/questions/35756682/getting-csv-sniffer-to-work-with-quoted-values), which worked for me:
Old:
`dialect = csv.Sniffer().sniff(csv_file.read(1024))`
New
`dialect = csv.Sniffer().sniff(csv_file.readline())`

kasbah, 2016-11-25:
@libbil, yes, waiting for this change to be merged in with #18

libbil, 2016-11-25:
great, well done

closed 2016-11-28 by xesscorp

gabedimafd, 2023-01-27:
I got this with latest version:
```/opt/homebrew/Cellar/csvkit/1.1.0/libexec/lib/python3.11/site-packages/agate/table/from_csv.py:70: RuntimeWarning: Error sniffing CSV dialect: Could not determine delimiter
/opt/homebrew/Cellar/csvkit/1.1.0/libexec/lib/python3.11/site-packages/agate/utils.py:274: UnnamedColumnWarning: Column 0 has no name. Using "a".
/opt/homebrew/Cellar/csvkit/1.1.0/libexec/lib/python3.11/site-packages/agate/utils.py:274: UnnamedColumnWarning: Column 1 has no name. Using "b".

PR #18 (2016-11-28): Increase the csv sniff size to the whole file
This fixes #17 by always passing the whole file to the CSV sniffer.
commit 9a9bcf1b (2016-11-28): Increase the csv sniff size to the whole file

This fixes #17 by always passing the whole file to the CSV sniffer.