devbisme/KiField #27: AttributeError: 'NoneType' object has no attribute 'lower'

antonbilbaeno, 2017-04-03:
This isn't exactly an issue. Just wondering if this error is common and if you might know off the top of your head some idea about what might be causing it?
Thanks,
Anton

ab@ab-mint1604 ~/src/folktek/official/13_channel/13_channel_v4 $ kifield -x 13_channel_v4.xlsx -i 13_channel_v4.sch --recurse

from shell:
> Traceback (most recent call last):
>   File "/usr/local/bin/kifield", line 9, in <module>
>     load_entry_point('kifield==0.1.5', 'console_scripts', 'kifield')()
>   File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.5-py2.7.egg/kifield/__main__.py", line 154, in main
>     recurse=args.recurse)
>   File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.5-py2.7.egg/kifield/kifield.py", line 1030, in kifield
>     insert_part_fields(part_fields_dict, insert_filenames, recurse)
>   File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.5-py2.7.egg/kifield/kifield.py", line 1014, in insert_part_fields
>     insertion_functions[f_extension](part_fields_dict, f, recurse)
>   File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.5-py2.7.egg/kifield/kifield.py", line 820, in insert_part_fields_into_sch
>     if unquote(f['name']).lower() == field_name.lower():
> AttributeError: 'NoneType' object has no attribute 'lower'

xesscorp, 2018-01-04:
Apologies for the delay. I can't produce the error without your input schematic. I'll close this issue. Re-open it if it still affects you and attach the schematic file causing the problem.

closed 2018-01-04 by xesscorp

poleguy, 2019-02-04:
I'm getting a similar error.

```
 
TypeError: descriptor 'lower' requires a 'str' object but received a 'NoneType'

Traceback (most recent call last):
  File "/usr/bin/kifield", line 9, in <module>
    load_entry_point('kifield==0.1.13', 'console_scripts', 'kifield')()
  File "/opt/anaconda/anaconda3/lib/python3.6/site-packages/kifield-0.1.13-py3.6.egg/kifield/__main__.py", line 148, in main
    backup = not args.nobackup)
  File "/opt/anaconda/anaconda3/lib/python3.6/site-packages/kifield-0.1.13-py3.6.egg/kifield/kifield.py", line 1211, in kifield
    insert_part_fields(part_fields_dict, insert_filenames, recurse, group_components, backup)
  File "/opt/anaconda/anaconda3/lib/python3.6/site-packages/kifield-0.1.13-py3.6.egg/kifield/kifield.py", line 1195, in insert_part_fields
    insertion_functions[f_extension](part_fields_dict, f, recurse, group_components, backup)
  File "/opt/anaconda/anaconda3/lib/python3.6/site-packages/kifield-0.1.13-py3.6.egg/kifield/kifield.py", line 878, in insert_part_fields_into_csv
    wb = insert_part_fields_into_wb(part_fields_dict, wb)
  File "/opt/anaconda/anaconda3/lib/python3.6/site-packages/kifield-0.1.13-py3.6.egg/kifield/kifield.py", line 772, in insert_part_fields_into_wb
    ref_col, _ = find_header_column(headers, id_label)
  File "/opt/anaconda/anaconda3/lib/python3.6/site-packages/kifield-0.1.13-py3.6.egg/kifield/kifield.py", line 319, in find_header_column
    lbl_match = lc_get_close_matches(lbl, header_labels, 1, 0.0)[0]
  File "/opt/anaconda/anaconda3/lib/python3.6/site-packages/kifield-0.1.13-py3.6.egg/kifield/kifield.py", line 310, in lc_get_close_matches
    lc_possibilities = [str.lower(p) for p in possibilities]
  File "/opt/anaconda/anaconda3/lib/python3.6/site-packages/kifield-0.1.13-py3.6.egg/kifield/kifield.py", line 310, in <listcomp>
    lc_possibilities = [str.lower(p) for p in possibilities]
TypeError: descriptor 'lower' requires a 'str' object but received a 'NoneType'

```
I'm guessing, but it seems to not like it if you try to extract a field that doesn't exist on every part.
If that's the case it would seem to be a very basic bug.

I have MPN and VPN fields, and VPN is not on every part:
```
 'U201': {'MPN': 'XC123454544 (Xilinx)'},
 'U4': {'MPN': '5P1234 (IDT)'},
 'U501': {'MPN': 'MX22345645 (Macronix) 1024345-ND (Digikey)',
          'VPN': '10243546-ND'},
 'X1': {'MPN': '123456878'}}
```

I can't share the whole schematic... maybe that's enough to reproduce the bug?