devbisme/KiField #28: File not found error with recurse option when the schematic is not located in the current working directory

kh90909, 2017-05-10:
The [recursive extraction code](https://github.com/xesscorp/KiField/blob/master/kifield/kifield.py#L495-L501) does not attach any path to sub-schematic filenames extracted from the top-level file. Thus, it fails to find those files if the schematics are not in the current working directory. The error is also somewhat cryptic due to the way it's [handled](https://github.com/xesscorp/KiField/blob/master/kifield/kifield.py#L683-L684). It states the top-level schematic file was not found, but of course, it must exist, otherwise it wouldn't have got this far.

Naively, the following change appears to fix the issue. It works for me, but I'm not familiar enough with how KiCad hierarchical schematics are implemented to know if this is foolproof (e.g. are absolute paths allowed for sub-schematics?)

```
diff --git a/kifield/kifield.py b/kifield/kifield.py
index a67d8be..07c9d8c 100644
--- a/kifield/kifield.py
+++ b/kifield/kifield.py
@@ -496,7 +496,7 @@ def extract_part_fields_from_sch(filename, inc_field_names=None, exc_field_names
         for sheet in sch.sheets:
             for field in sheet.fields:
                 if field['id'] == 'F1':
-                    sheet_file = unquote(field['value'])
+                    sheet_file = os.path.join(os.path.dirname(filename),unquote(field['value']))
                     part_fields_dict.update(extract_part_fields_from_sch(sheet_file, inc_field_names, exc_field_names, recurse, depth+1))
                     break
```

4b11b4, 2018-08-07:
I also have this issue. Top level schematic is not found, although it certainly exists in the directory.
KiField 0.1.12

kasbah, 2018-08-07:
Could you make a pull-request with this fix and (ideally) a test that checks for this behaviour?

4b11b4, 2018-08-09:
@kasbah 

Could you give a brief description of how the test would work?

4b11b4, 2018-08-09:
I created a pull request, feel free to comment there.

closed 2018-10-28 by xesscorp