devbisme/KiField #31: enhancement: combine references with identical fields into a single line

ppelleti, 2017-09-15:
It would be great to have an option that would combine references with identical fields into a single line.  For example,

```
R3,,Resistors_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P7.62mm_Horizontal,470
R4,,Resistors_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P7.62mm_Horizontal,470
R5,,Resistors_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P7.62mm_Horizontal,470
R6,,Resistors_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P7.62mm_Horizontal,470
R7,,Resistors_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P7.62mm_Horizontal,10k
R8,,Resistors_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P7.62mm_Horizontal,10k
```

would become:

```
R3;R4;R5;R6,,Resistors_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P7.62mm_Horizontal,470
R7;R8,,Resistors_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P7.62mm_Horizontal,10k
```

That would make it easier to add things like manufacturer part number, which would be the same for identical components, rather that having to enter it separately for each reference.

kasbah, 2017-09-15:
I think you are looking for the `--group` option.

```
  --group, -g           Group components with the same field values into
                        single lines when inserting into a spreadsheet or
                        CSV/TSV. (Default is to have one component per line)
```

ppelleti, 2017-09-15:
Awesome, thank you!  That seems to be missing from the [usage page](https://kifield.readthedocs.io/en/latest/usage.html).  :)

closed 2017-09-18 by xesscorp