devbisme/KiField #32: AttributeError: 'long' object has no attribute 'startswith'

ppelleti, 2017-09-15:
I am getting the following error when trying to insert fields from an .xlsx back into the schematic:

```
whiteandnerdy:NeoESP ppelleti$ kifield -x NeoESP.xlsx -i NeoESP.sch 
Traceback (most recent call last):
  File "/usr/local/bin/kifield", line 8, in <module>
    load_entry_point('kifield==0.1.7', 'console_scripts', 'kifield')()
  File "/Library/Python/2.7/site-packages/kifield-0.1.7-py2.7.egg/kifield/__main__.py", line 157, in main
    recurse=args.recurse)
  File "/Library/Python/2.7/site-packages/kifield-0.1.7-py2.7.egg/kifield/kifield.py", line 1154, in kifield
    insert_part_fields(part_fields_dict, insert_filenames, recurse, group_components)
  File "/Library/Python/2.7/site-packages/kifield-0.1.7-py2.7.egg/kifield/kifield.py", line 1138, in insert_part_fields
    insertion_functions[f_extension](part_fields_dict, f, recurse, group_components)
  File "/Library/Python/2.7/site-packages/kifield-0.1.7-py2.7.egg/kifield/kifield.py", line 918, in insert_part_fields_into_sch
    if field_value.startswith(INVIS_PREFIX):
AttributeError: 'long' object has no attribute 'startswith'
```

I'm on Mac OS X 10.9.5, and I installed kifield as recommended with `easy_install kifield`.

```
whiteandnerdy:NeoESP ppelleti$ python --version
Python 2.7.5
```

ppelleti, 2017-09-15:
I think maybe the problem is that some of my manufacturer part numbers contain only digits and no letters, so it's thinking they are an integer instead of a string?

I was able to fix the problem by saving my .xlsx as a .csv in LibreOffice, and then running kifield on my .csv.

Here is the offending .xlsx file:

[NeoESP.xlsx](https://github.com/xesscorp/KiField/files/1307612/NeoESP.xlsx)

xesscorp, 2017-09-18:
Thanks. I'll look into it.

xesscorp, 2017-09-22:
I need the NeoESP.sch file that goes with the NeoESP.xlsx file. Otherwise I can't test my fix.

ppelleti, 2017-09-22:
Sure, here is the `NeoESP.sch` file.  Github didn't want to let me attach the `.sch` file directly, so I put it inside a `.zip` file to make Github happy:

[NeoESP.zip](https://github.com/xesscorp/KiField/files/1323345/NeoESP.zip)

ppelleti, 2017-09-22:
That previous attachment was the `.sch` after I'd put the `mpn` field back into it.  You probably want it before the `mpn` field was added, since that's what you're testing.  So here it is without the `mpn` field.  I also threw in `NeoESP-cache.lib`, too, in case you need it.

[NeoESP.zip](https://github.com/xesscorp/KiField/files/1323354/NeoESP.zip)

closed 2017-09-23 by xesscorp