devbisme/KiField #37: "Unable to write to file"

01tot10, 2018-01-16:
First time opening an issue, hope I'm doing this right! 

Running Ubuntu Gnome 17.04 Codename: zesty

After installing my operating system from scratch, I get the following error when trying to run kifield
"kifield -x sheetname.sch -i test.csv"
-> _Unable to write to file: test.csv._
"kifield --help" gives a proper output (help documentation).
I can also "touch test.sch".
I also tried "sudo kifield -x sheetname.sch -i test.csv" with a similar issue.

It might be that I'm just doing something stupid here, but can't figure out what.

Installed kifield with "sudo easy_install kifield". Installed succesfully:
_Searching for kifield_
_Best match: kifield 0.1.11_
_Processing kifield-0.1.11-py2.7.egg_
_kifield 0.1.11 is already the active version in easy-install.pth_
_Installing kifield script to /usr/local/bin_

_Using /usr/local/lib/python2.7/dist-packages/kifield-0.1.11-py2.7.egg_
_Processing dependencies for kifield_
_Finished processing dependencies for kifield_

Any suggestions? 

Thank you in advance!

kasbah, 2018-01-16:
Does this happen with any `.sch` file? Would you be able to share a `.sch` file that this happens with?
Could you make sure to remove the `test.csv` to run the command? Have you tried with `.ods|.xlsx|.tsv` output?

You could you try installing with pip and in a virtualenv:

```
sudo apt install python-virtualenv pip
virtualenv venv
source venv/bin/activate
pip install kifield
kifield -x sheetname.sch -i sheetname.csv
```

boerontour, 2018-01-16:
I have the same problem. Both installation methods (easy_install and pip in vitualenv) on Ubuntu 16.04 give the same result: "Unable to write file: test.csv". Same result for other output file types and other .sch input files.

kasbah, 2018-01-16:
I found the problem and sent a pull-request. There is a problem in the backup routine. You can work around the problem, at least with .csv files, by first doing `echo ',' > test.csv`.

boerontour, 2018-01-16:
Thanks for your fast response. Creating a file is a work around for the problem. Creating an empty .xlsx with Excel works also.

01tot10, 2018-01-16:
Tried _echo ',' > test.csv_ and it fixed the issue! Thanks everyone for the fast reactions! 

Also, can't express how excited I am getting an answer to my first issue ever within the first day!
And even more one that solves my troubles. Good times!

kasbah, 2018-01-16:
Glad to help. The `--nobackup` flag is another workaround where you don't have to create a file first.

PR #39 (2018-01-18): Make sure there is a file before backing up
Fixes #37
closed 2018-01-18 by xesscorp