devbisme/KiField #38: error messages

Andrewdwork, 2018-01-16:
Hi. I am running win7 64 bit.  Python is V3.6.4:d48eceb, Dec 19 2017 build.
I have successfully installed KiField (I think). I have a Kicad schematic that I have put in C:\Kifield\ AGC_Preamp.sch  for ease of access. I have made BOMs from this using the PCB package. All the devices have been back annotated from the PCB fabrication outputs BOM tab (so it should work).
I run Kifield -x C:\Kifield\AGC_Preamp.sch  -i C:\Kifield\AGC_Preamp.csv from the command prompt, then when I get the spelling and spacing right it runs, but exits with a "could not determine delimiter" error at csv.py line 202.
If i then ask it to produce an xlsx file by changing the last extension to .xlsx,
I get errors in excel.py line 121 where it looks for a ZIP file. The ultimate error is that zipfile.py line 1175 decides that the file is not a ZIP archive, which is true.
The traceback for both runs is attached as a jpg.
Have I not done something, or missed some steps out?
Any help would be appreciated.
[image]

kasbah, 2018-01-16:
Hmm.. maybe you have some bad files lying around and KiField is trying to insert into them. Could you make sure to remove any *.csv and *.xlsx files and try again?

Andrewdwork, 2018-01-22:
Hi. Sorry for the delay, . Your reply had vanished into my Spam folder for some reason.

I deleted the existing CSVs and XLS files, and ran the program again. 
The message was Cannot write to *.csv (just too idle to fill in full title - see picture).
Created an empty *.csv file. Same message.
Tried for xlsx file, same message. Created blank txt file, renamed extension to xlsx, same message again.
Tried the -w option. Still get "cannot write to file message".

The only files in the Kifield folder are a blank csv, blank xlsx and the .sch file.

Screen shot attached.
[Document1.pdf](https://github.com/xesscorp/KiField/files/1651584/Document1.pdf)

closed 2018-01-22 by Andrewdwork