devbisme/KiField #40: Type errors

ChrisGammell, 2018-02-07:
Have done multiple imports/exports in the past without issue. It appears I've borked something and am wondering if there's a way to track down where I may have mistakenly inserted a bad character. Is it safe to assume that it was due to something I inserted from this error message?

Can no longer do an export or an import.

```
Traceback (most recent call last):
  File "/usr/local/bin/kifield", line 9, in <module>
    load_entry_point('kifield==0.1.12', 'console_scripts', 'kifield')()
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.12-py2.7.egg/kifield/__main__.py", line 148, in main
    backup = not args.nobackup)
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.12-py2.7.egg/kifield/kifield.py", line 1207, in kifield
    insert_part_fields(part_fields_dict, insert_filenames, recurse, group_components, backup)
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.12-py2.7.egg/kifield/kifield.py", line 1191, in insert_part_fields
    insertion_functions[f_extension](part_fields_dict, f, recurse, group_components, backup)
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.12-py2.7.egg/kifield/kifield.py", line 866, in insert_part_fields_into_csv
    wb, dialect = csvfile_to_wb(filename)
  File "/usr/local/lib/python2.7/dist-packages/kifield-0.1.12-py2.7.egg/kifield/kifield.py", line 210, in csvfile_to_wb
    setattr(dialect, attr, bytes(a))
  File "/usr/local/lib/python2.7/dist-packages/future-0.16.0-py2.7.egg/future/types/newbytes.py", line 93, in __new__
    raise TypeError('unicode string argument without an encoding')
TypeError: unicode string argument without an encoding
```

ChrisGammell, 2018-02-07:
I saw #39 had a similar problem and verified that I could export again without issue, so that took care of that. However, I turn around and try to immediately import what i just export and it gives the same errors as above. This is the command I'm using

```kifield -g -x <testfile>.csv -i <testfile>.sch --overwrite```

kasbah, 2018-02-07:
Hi Chris!  Can you share the files? Can you try it with Python3?

ChrisGammell, 2018-02-07:
Probably can't share the files because it's for work, but happy to try it with python3! How do I make the switch? Did I miss that somewhere in the docs?

kasbah, 2018-02-07:
What OS are you on, how did you install kifield? Often it's just a matter of getting python3 installed and doing `pip3 install kifield` or even  `pip3 install git+https://github.com/xesscorp/KiField` if you want the latest version to avoid the backup bug.

ChrisGammell, 2018-02-07:
On Ubuntu 16.04, I used ```easy_install kifield```

Just removed the old package, did what you suggested and got this

```
Traceback (most recent call last):
  File "/usr/local/bin/kifield", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2927, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2913, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2940, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 635, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 943, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 829, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'kifield==0.1.12' distribution was not found and is required by the application
````

ChrisGammell, 2018-02-07:
Hrm, did the easy_install again and it's back to working. But not using Python3.

kasbah, 2018-02-07:
You are still on 2.7 ^__^

Try:

```
sudo pip uninstall kifield
sudo apt install python3-pip
sudo pip3 install git+https://github.com/xesscorp/KiField
```

ChrisGammell, 2018-02-07:
Ah, had to add the --upgrade tag for pip3 to pull in the required files.

That seemed to work! Thanks for all the help!

closed 2018-02-07 by ChrisGammell

tkstreet, 2018-04-23:
Trying to do this on windows and it's failing.  Python doesn't play well with Cygwin at all.  It keeps trying to mix Windows and POSIX environment variables:

```
E:\tstreet\workspace\AKDP\AKDP_kicad\mother_board\RevE1>pip3 install git+https://github.com/xesscorp/KiField
Collecting git+https://github.com/xesscorp/KiField
  Cloning https://github.com/xesscorp/KiField to c:\users\tstreet\appdata\local\temp\pip-req-build-x96ct_e8
fatal: Invalid path '/cygdrive/e/tstreet/workspace/AKDP/AKDP_kicad/mother_board/RevE1/C:\Users\tstreet\AppData\Local\Temp\pip-req-build-x96ct_e8': No such file or directory
```

tkstreet, 2018-04-23:
Nevermind. I nuked KiField in both Python 2 and Python 3, then installed it on Python 3 using pip3 and now it is working great. Thanks!