devbisme/KiField #43: kifield start error message

szg0000, 2018-07-12:
I have error message while running kifield:

```
Traceback (most recent call last):
  File "kifield.py", line 50, in <module>
    from .sch import Schematic
ModuleNotFoundError: No module named '__main__.sch'; '__main__' is not a package
```

kasbah, 2018-07-12:
What version of python are you running? Have you tried it with Python3?

szg0000, 2018-07-12:
Same problem with `Python 3.6.2rc1` and `Python 3.7.0b2`.

kasbah, 2018-07-12:
And python 2.7?

kasbah, 2018-07-12:
Could you give more details that would allow me to reproduce this issue please. Can you share the file and the exact command you are running please.

szg0000, 2018-07-12:
With python2.7 have broken package dependecies:

```
root@debian:/home/user/KiField/kifield# pip install future
Requirement already satisfied: future in /usr/local/lib/python2.7/dist-packages (0.16.0)
pycollada 0.4 has requirement python-dateutil==1.5, but you'll have python-dateutil 2.6.1 which is incompatible.
```

szg0000, 2018-07-12:
Thanks, works now.

closed 2018-07-12 by szg0000