devbisme/KiField #44: example needed, how making fields visible or invisible

szg0000, 2018-07-12:
After running
`kifield -x schematicfile.sch -i schematicfile.csv`

I added `[I]` to the header to make the refs invisible.
```
[I]Refs,datasheet,footprint,value
D1,~,Diode_SMD:D_SMC,SMCJ24A
D10,~,Diode_SMD:D_MiniMELF,5V
D11,~,Diode_SMD:D_MiniMELF,5V
D12,~,Diode_SMD:D_MiniMELF,5V
D13,~,Diode_SMD:D_MiniMELF,5V
```
But after 
`kifield -x schematicfile.csv -i schematicfile.sch` the refs are visible in the schematic and the PCB board also.

Can someone give a working example how to make all refs invisible on a PCB board?

kasbah, 2018-07-12:
Does it say somewhere that adding `[I]` will make it invisible?

Check `kifield -h`: 

```
  --fields name|/name|~name [name|/name|~name ...], -f name|/name|~name [name|/name|~name ...]
                        Specify the names of the fields to extract and insert.
                        Place a '/' or '~' in front of a field you wish to
                        omit. (Leave blank to extract/insert *all* fields.)
```

kasbah, 2018-07-12:
Ah [just found it](https://xesscorp.github.io/KiField/docs/_build/singlehtml/index.html#making-fields-visible-or-invisible). Didn't know about that. But I think that only makes it "invisible" in the schematic, i.e. it's still there but you can't see it when looking at the schematic.

`--fields` option above should do what you want.

szg0000, 2018-07-12:
Can you give a try and give a working command example how to do this?
Can you create a small schematic with a few elements and PCB, and try to make all Refs invisible on schematic and on the PCB also.

kasbah, 2018-07-12:
Sorry, don't have time for that right now. If you can make the example I can try and confirm the issue at least.

szg0000, 2018-07-13:
> Maybe try a space after the [I]?

Did not help. Any new idea?

kasbah, 2018-07-18:
`I` without the square brackets?

xesscorp, 2018-07-19:
KiField doesn't process any changes to part references so their visibility/invisibility can't be changed. Immutability of part references was baked into KiField from the start, so getting it to change anything about a reference is difficult. I spent two hours trying to make it work, but failed.