devbisme/KiField #46: Feature request: minimize diffs

ppelleti, 2018-09-25:
One of the reasons I don't use kifield as often as I otherwise might is that it creates large, unnecessary diffs.  This makes it harder to tell what has changed in a particular commit.

There are at least three ways kifield creates larger diffs than necessary:

* Changes in whitespace on an individual line: In certain places, KiCad uses two spaces between tokens instead of one.  Kifield always puts just one space between tokens.
* Changes in field numbers: If my original file has F4 as "MPN", F5 as "Description", and F6 as "Digi-Key_PN", Kifield might change it to be F4 as "Description", F5 as "Digi-Key_PN", and F6 as "MPN".
* Ordering within the file: Kifield moves all components to come first in the file, and everything else (wires, junctions, etc.) to come after all the components.

xesscorp, 2018-09-28:
You're right that schematic diffs can get large.

The solution for point #1 is to disable whitespace compares.

Regarding point #2, KiCad doesn't seem to have any standardization as to which of the field numbers is associated with each type of information (except for F0-F2). I've seen the same field number used for different types of information in a single file. KiField would have to remember which field number went with each individual field rather than trying to make all the fields consistent as it does now.

As to point #3, the order of the internal structures of the schematic file isn't strictly specified. I'm using the order I've seen in my schematic files and which was given in a Python script I got from https://github.com/KiCad/kicad-library-utils/tree/master/sch. Maybe the order has changed with V5. I don't know. (I'm sure it will change completely with V6 when they move to using S-expressions.) In any event, the order of sections would be easy to change in the `save()` method of the `sch.py` file. However, the order of the *individual* wires, junctions, components, etc. within each section may not be preserved, so this may not be much of a solution.

ppelleti, 2018-09-28:
> Maybe the order has changed with V5. I don't know.

I don't know about v5, but my experience with 4.0.7 is that everything just appears in the file in the order in which it was created, so there are no "sections".  The wires, junctions, components, etc. are intermixed.

nhatkhai, 2018-11-12:
The parser do a sort, with cause everything. In 4.7.x they aren't save back in the file that way - which actual better compare to blindly sort everything, an make more sense when a human look at the text file of .lib, .sch ..... I had mention this on kicad bug report too.

nhatkhai, 2019-01-23:
I'm sort of solving this issue in my script using my own iterator parser: [nhatkhai/kicad_scripts](https://github.com/nhatkhai/kicad_scripts/blob/master/bom_plugins/bom2sch.py)