devbisme/KiField #50: KiField fails to write new fields which do not exist in the .sch file

4b11b4, 2019-01-31:
Maybe my understanding of the --overwrite functionality is incorrect.

In a .sch file, a component is defined as such:
$Comp
L Device:R_Small R73
U 1 1 5B382555
P 5750 2400
F 0 "R82" H 5780 2420 50 0000 L CNN
F 1 "100k" H 5780 2360 50 0000 L CNN
F 2 "passive:0603_pad-0.95W-0.7L" H 5750 2400 50 0001 C CNN
F 3 "~" H 5750 2400 50 0001 C CNN
 1 5750 2400
 0 -1 -1 0
$EndComp

However, I have new fields which I have created via the "Field Name Templates". One of these fields is called "Power". All of my new fields are properly shown when pressing 'E' on a component in Eeschema (these new fields are indeed blank).

If I manually put a "-" into the Power field via editing a component with 'E', then this change is reflected in the schematic as:

$Comp
L Device:R_Small R73
U 1 1 5B381EC3
P 3000 2950
F 0 "R73" H 3030 2970 50 0000 L CNN
F 1 "100k" H 3030 2910 50 0000 L CNN
F 2 "passive:0603_pad-0.95W-0.7L" H 3000 2950 50 0001 C CNN
F 3 "~" H 3000 2950 50 0001 C CNN
F 4 "-" H 3000 2950 50 0001 C CNN "Power"
 1 3000 2950
 -1 0 0 1
$EndComp

So, by default KiCad is not writing these fields which are empty. This may be the intended functionality. I would assume that any fields which are displayed when pressing 'E' in Eeschema would be written.
I have filed a bug here: https://bugs.launchpad.net/kicad/+bug/1814166
 
I seem to have remembered being able to successfully write new fieldsvia the "--overwrite" flag previously. Maybe I am mistaken.

4b11b4, 2019-02-01:
I wrote a .sh script which uses sed which wrote a default value to all of these fields.
When I edit a component using 'E' in Eeschema, I can see that all of my new fields have the value "-".

However, KiField is still failing to write new values to these fields.

4b11b4, 2019-02-01:
--overwrite and --recurse flags are set.
with --debug 10 I can see that the proper values are extracted from the .xlsx file
piping the whole command to grep for one of these components where it doesn't seem to be writing (it does write to some...)
reveals that it skips the "Updating" part for these components

4b11b4, 2019-02-01:
After some "print-line" and grep debugging it appears this component doesn't exist at all in:
sch.components > insert_part_fields_into_sch function.

4b11b4, 2019-02-01:
Looking at a component that is written:
$Comp
L Device:R_Small R272
U 1 1 5C447076
P 4900 6150
F 0 "R272" H 4930 6170 50 0000 L CNN
F 1 "100k" H 4930 6110 50 0000 L CNN
F 2 "passive:0603_pad-0.95W-0.7L" H 4900 6150 50 0001 C CNN
F 3 "" H 4900 6150 50 0001 C CNN
F 4 "Thick" H 4930 6270 50 0001 C CNN "Composition"
F 5 "0.0625" H 4930 6270 50 0001 C CNN "Power"
F 6 "Yes" H 4930 6270 50 0001 C CNN "Sub"
F 7 "0.05" H 4930 6270 50 0001 C CNN "Tolerance"
	1    4900 6150
	1    0    0    -1  
$EndComp

And one that is not:
$Comp
L Device:R_Small R75
U 1 1 5B3840AF
P 5000 3500
F 0 "R75" H 5030 3520 50  0000 L CNN
F 1 "100k" H 5030 3460 50  0000 L CNN
F 2 "passive:0603_pad-0.95W-0.7L" H 5000 3500 50  0001 C CNN
F 3 "~" H 3000 2950 50  0001 C CNN
F 4 "-" H 3000 2950 50  0001 C CNN "Power"
F 5 "-" H 3000 2950 50  0001 C CNN "Description"
F 6 "-" H 3000 2950 50  0001 C CNN "MFR"
F 7 "-" H 3000 2950 50  0001 C CNN "MPN"
F 8 "-" H 3000 2950 50  0001 C CNN "URL"
F 9 "-" H 3000 2950 50  0001 C CNN "Sub"
F 10 "-" H 3000 2950 50  0001 C CNN "Installed"
F 11 "-" H 3000 2950 50  0001 C CNN "Note"
F 12 "-" H 3000 2950 50  0001 C CNN "Composition"
F 13 "-" H 3000 2950 50  0001 C CNN "Dielectric"
F 14 "-" H 3000 2950 50  0001 C CNN "Tolerance"
F 15 "-" H 3000 2950 50  0001 C CNN "Volt."
	1    5000 3500
	-1   0    0    1   
$EndComp

...no obvious differences...

4b11b4, 2019-02-01:
It may be that:
in kifield.py, line 909
    try:
        sch = Schematic(filename)

is not getting all of my components in the schematic. I have re-exported the netlist. Still nothing.

4b11b4, 2019-02-01:
After some more digging,
KiField is specifically not loading schematics which are nested more than once.

my current setup is:
proj/
  componentboard/
    comp.proj
    comp.sch
  sch/
    cv/
      cv1.sch
      cv2.sch
    led/
      led1.sch
      led2.sch

The problem occurs when comp.sch has a block which links to cv1.sch for example.
The path for this block (relative to the KiCad Project) is: "../sch/cv/cv1.sch"

The way KiCad deals with paths, since this first block is located at "../sch/cv", a block INSIDE this block now is already at this path ("../sch/cv"), and **thus**, if I created a block for cv2 _inside_ of cv1, the path for cv2 would simply be: "cv2.sch". <--- **THIS** is where KiCad is failing. It is reading these filenames relative to where the main schematic is located (where you run the command).

Obviously, "cv2.sch" doesn't exist in this main directory, and as such when it tries to read this filename it fails.

4b11b4, 2019-02-01:
Wayne decided that the path of "subsheets" would be relative to the block above, making them more portable.

Bug I filed: https://bugs.launchpad.net/kicad/+bug/1814166
* this is irrelevant as this is probably the intended behavior...

Other bugs related to these issues:
https://bugs.launchpad.net/kicad/+bug/1735982
https://bugs.launchpad.net/kicad/+bug/1667623

4b11b4, 2019-02-01:
Is there any way for KiField to determine if a sheet is nested within another sheet? and pre-pend the path?

4b11b4, 2019-02-01:
I have fixed this issue by prepending a path to the sheet_file variable.

4b11b4, 2019-02-01:
#51

xesscorp, 2019-02-08:
I simplified your fix and ran the tests. Yo can check to see if it still works on your schematic. Thanks for the help!

closed 2019-02-08 by xesscorp