devbisme/KiField #58: Representing inches with a quote mark causes causes import error

RobertHorst, 2020-07-22:
Some part descriptions represent inches with a double quote (e.g. 0.91" OLED).  When this is imported back to KiCAD, attempting to open the schematic causes an error.  The schematic can be opened only after using a text editor to remove the quote from the .sch file.  Single quotes likely have the same problem.

Best would be to escape single and double quotes.  

If that is difficult or causes other problems, it would be OK to give an error message and refuse to import data with quotes in the data fields.

xesscorp, 2020-07-24:
I pushed a new commit to the `development` branch that may fix this issue. You can install it with
```bash
pip install git+https://github.com/xesscorp/KiField@development
```
Let me know if this works for you. If not, please send me some files that illustrate the problem.

RobertHorst, 2020-07-25:
No, this did not fix the problem.

The attached files show the problem.  The .bak is the before version of the schematic. KiField was used to create Test_field.xlsx.  That was edited with Excel with a description field having a double quote and the modified Excel file saved to Test_field2.xlsx.  KiField uploaded that back to “Test schematic for KiField.sch.”  When you attempt to open it with KiField, It fails with

Error loading schematic file “/Users/.../KiCad test project/Test schematic for KiField.sch”.
expected single character token in input/source
"/Users/.../Test schematic for KiField.sch"
line 24, offset 30

Looking at that this file with an editor shows unbalanced quotes on line 24.

— Bob

xesscorp, 2020-07-25:
I didn't get any attached files, but I was able to recreate your problem. Go ahead and try the latest version in the development branch.

RobertHorst, 2020-07-26:
Just tried it with my test file, and it worked correctly.  

After running KiField to insert a quote mark in the xlsx file, I can see the  escaped quote  with a text editor.  Opening it with KiCad, it shows up as a normal quote mark in the description field as it should.

Thanks for the quick response.

— Bob

xesscorp, 2020-07-26:
Sounds good!

closed 2020-07-27 by xesscorp