devbisme/KiField #59: Add quantity column

ryanfobel, 2020-11-17:
When generating a BOM from a schematic, you can group components with the same value using the ```-g``` or ```--group``` command line switch, but in this case, I would expect a "Quantity" column to include the number of components. Is there a way to generate this automatically?

psheph, 2021-05-27:
I'd be up for taking a whack at adding this. Is there a preference for branch naming? Do I need to be added as a contributor, or is there a different way for me to share proposed changes?

xesscorp, 2021-05-27:
1. Fork this repo.
2. Make changes to the **development** branch.
3. Issue a PR.

I'm not sure what this feature actually accomplishes since you can see the list of references in each group. The only time you would need the quantity is if there were a large number of identical parts so the reference list would be too large for the spreadsheet cell. But do you really care about the exact number?

psheph, 2021-05-27:
>But do you really care about the exact number?

Heck yeah! :joy:

I'm not sure you are aware of how many ways people want to use KiField! For example, our intern, @jerryhgoss, recently used KiField as a part of a project to download part metadata from Octopart, and then automatically push these properties back into the schematic. (this doesn't require the qty field, of course) https://github.com/jerryhgoss/Octopart_KiCad_Integration

The motivation for exporting qty is to generate better BOMs for sending to CMs. The native export capability of KiCad v5 won't group by properties such as "DNP", but KiField will. The fact that KiField uses - notation for lists of consecutive reference designators makes for a beautiful BOM, but makes it harder to count the qty in a script or Excel. I've been poking around, and I think I have an idea where to stick the accumulator during grouping. At least for one or two file formats.

xesscorp, 2021-05-29:
I like your first application; that's what KiField is for.

I'm a bit leery of using KiField in a BOM-type application. That kind of happened with my KiCost app and I don't think it was a good idea. Still, it's FOSS so go ahead and try! It wouldn't be the first time I was wrong.

I'm doing some work on KiField to support KiCad V6 schematic and library formats, so be aware the source is changing. It may require some manual effort on my part to reconcile your changes and mine.