devbisme/KiField #60: Strange behavior generating .csv/.xlsx from hierarchical schematic with grouped components

scandey, 2021-02-12:
I've been attempting to use KiField on a sizable, pre-existing, schematic with lots of duplicated hierarchical sheets and I've been running into some strange issues. This is a fairly isolated one though, so I'm starting with it.

Using the hier_test schematic in tests/integration and running `kifield -x hier_test.sch -i test.csv -r -g -w -d 10`
results in the expected:
```
Refs,datasheet,footprint,new_field,value
"Y1, Y2",my_datasheet,my_footprint,my_field,Crystal
```
I edited test.csv to add a field and value:
```
Refs,datasheet,footprint,new_field,value,newfield
"Y1, Y2",my_datasheet,my_footprint,my_field,Crystal,newvalue
```
Running `kifield -x test.csv -i hier_test.sch -r -g -w -d 10`
results in the components both displaying the correct new field and value as expected when I open the schematic in Kicad.

However, if I then try to re-export the fields to the same csv file with the exact same command in the first line, the grouping seems to disappear.
text.csv:
```
Refs,datasheet,footprint,new_field,value,newfield
Y1,my_datasheet,my_footprint,my_field,Crystal,newvalue
```

Is this expected behavior? It seems like I should get back out exactly what I put in between inserting into the schematic and then extracting immediately back out. 
Additionally, once I've triggered the behavior, deleting the file and running `kifield -x hier_test.sch -i test.csv -r -g -w -d 10` twice in a row also results in de-grouping. 

Behavior occurs regardless of the -w overwrite flag and regardless of .csv or .xlsx output.

There are some subtle differences in the debug messages, specifically a slightly difference in the Exploding sequence.

First run insertion debug log:
```
Inserting extracted fields into files ['test.csv'].
Inserting fields into test.csv.
Inserting extracted fields into CSV file test.csv.
Converting CSV file test.csv into an XLSX workbook.
Header on row 1: ['Refs', 'datasheet', 'footprint', 'new_field', 'newfield', 'value'].
Found Refs on header column 1.
Exploding Y1 => ['Y1'].
Exploding Y2 => ['Y2'].
Exploding Y1 => ['Y1'].
Updating Y1 field value from None to Crystal
Updating Y1 field footprint from None to my_footprint
Updating Y1 field datasheet from None to my_datasheet
Updating Y1 field new_field from None to my_field
Updating Y1 field newfield from None to newvalue
Exploding Y2 => ['Y2'].
Updating Y2 field value from None to Crystal
Updating Y2 field footprint from None to my_footprint
Updating Y2 field datasheet from None to my_datasheet
Updating Y2 field new_field from None to my_field
Updating Y2 field newfield from None to newvalue
Converting an XLSX workbook and saving as CSV file test.csv.
```

Second run insertion debug log:
```
Inserting extracted fields into files ['test.csv'].
Inserting fields into test.csv.
Inserting extracted fields into CSV file test.csv.
Converting CSV file test.csv into an XLSX workbook.
Header on row 1: ['Refs', 'datasheet', 'footprint', 'new_field', 'newfield', 'value'].
Found Refs on header column 1.
Exploding Y1, Y2 => ['Y1', 'Y2'].
Exploding Y1, Y2 => ['Y1', 'Y2'].
Updating Y1 field value from Crystal to Crystal
Updating Y1 field footprint from my_footprint to my_footprint
Updating Y1 field datasheet from my_datasheet to my_datasheet
Updating Y1 field new_field from my_field to my_field
Updating Y1 field newfield from newvalue to newvalue
Updating Y2 field value from Crystal to Crystal
Updating Y2 field footprint from my_footprint to my_footprint
Updating Y2 field datasheet from my_datasheet to my_datasheet
Updating Y2 field new_field from my_field to my_field
Updating Y2 field newfield from newvalue to newvalue
Converting an XLSX workbook and saving as CSV file test.csv.
```

I'm not sure where to go from here though. I'm happy to run more tests if that would be useful! I'm also investigating odd behavior where some insertions to a complex schematic leads to the schematic files corrupting with no placement information following the footprint, but that's for a separate issue.

**Version info**

KiField 0.1.16 
Python 3.9.1
macOS 11.2
KiCad 5.1.9-0-10_14

KiCad detailed version info:
<details>

```
Application: KiCad
Version: (5.1.9-0-10_14), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.41.0
Platform: macOS Version 10.16 (Build 20D64), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
    Boost: 1.75.0
    OpenCASCADE Technology: 7.5.0
    Curl: 7.54.0
    Compiler: Clang 10.0.1 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

</details>

scandey, 2021-07-26:
I just caught this old issue happening again, where on extract from schematic and insert into an existing spreadsheet, only the first item in a group is included in the components list. 

I'm using recursive, grouped, no backups, and overwrite flags. While I normally increment a revision number in the filename of the working spreadsheet, I would assume that this combo of flags should allow for repeated use of the same spreadsheet filename without issues.

xesscorp, 2021-07-26:
Yes, I've been working on this one before releasing a new version of KiField.

xesscorp, 2021-07-26:
I think this is fixed now in the development branch.

scandey, 2021-07-27:
That worked for me! There's a strange side effect where the reference designators are now sorted oddly:
C1, C105, C106, C162, C168, C170, C179, C181, C182, C183, C195, C196, C197, C198, C199, C2, C200, C201, C203, C255, C257, C75, C76
Not a problem, just an oddity. That being said, I really appreciate the change to showing every component in the list, makes it much easier to search!

devbisme, 2021-07-27:
Yeah, I introduced an error (again!) in the way part references are collapsed. I'm fixing it now.

Can't you just omit the `-g` option if you want the part references all explicitly listed?

scandey, 2021-07-27:
(Personal opinion based on my workflow, not a functionality issue) 
I don't tend to use the ungrouped option since if I want to make a change to a component's fields, I always want to make it to all references of that component. I also produce BOMs with all of the reference designators for a single line item listed explicitly (C1, C2, C3, C8 rather than C1-C3, C8), though I realize that is down to preference. 

So while in the process of editing components, it is nice to be able to CTRL-F search for a reference and guarantee that I'll find it if it exists. If I pick a common component on the schematic, I tend to just search the Kifield spreadsheet for it by a single reference designator, but if it is in the middle of a range it won't appear in the search and I often have a tiny moment of panic before I remember to look for the range it could be in. 

If I were to add a feature request, it would be for a flag to not collapse parts references before output to csv/xlsx (or re-explode them after). If that sounds reasonable, I can open a new ticket for it (and potentially try doing it myself!)

devbisme, 2021-07-27:
You could add a `--no_range, -nr` option that disables hyphenated ranges and explicitly shows each component reference in a group.

closed 2023-07-30 by scandey