devbisme/KiField #76: [KiField Bug] Kifield not compatible with Kicad 7

dilydaly1, 2023-03-08:
**Describe the bug**
A clear and concise description of what the bug is.
Although I was able to read from the V7 library file (.kicad_sym) into an excel sheet the write back was unsuccessful. 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
See output from Kifield below

**Expected behavior**
A clear and concise description of what you expected to happen.
Normal execution
**Screenshots**
If applicable, add screenshots to help explain your problem.
Traceback (most recent call last):
  File "C:\Users\mlindsey\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main        
    return _run_code(code, main_globals, None,                                                                           
  File "C:\Users\mlindsey\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code                   
    exec(code, run_globals)                                                                                              
  File "C:\Users\mlindsey\PycharmProjects\pythonProject\venv\Scripts\kifield.exe\__main__.py", line 7, in <module>       
  File "C:\Users\mlindsey\PycharmProjects\pythonProject\venv\lib\site-packages\kifield\__main__.py", line 149, in main   
    kifield(                                                                                                             
  File "C:\Users\mlindsey\PycharmProjects\pythonProject\venv\lib\site-packages\kifield\kifield.py", line 1554, in kifield
    insert_part_fields(
  File "C:\Users\mlindsey\PycharmProjects\pythonProject\venv\lib\site-packages\kifield\kifield.py", line 1518, in insert_part_fields
    insertion_function(
  File "C:\Users\mlindsey\PycharmProjects\pythonProject\venv\lib\site-packages\kifield\kifield.py", line 1407, in insert_part_fields_into_lib_V6
    component.copy_field("Reference", field_name)
  File "C:\Users\mlindsey\PycharmProjects\pythonProject\venv\lib\site-packages\kifield\schlib.py", line 568, in copy_field
    id = find_by_key("id", dst_field["prop"])[0]
IndexError: list index out of range

**Desktop (please complete the following information):**
 - OS [type & version] windows10
 - Python version
 -Python 3.10.7 (tags/v3.10.7:6cc6b13, Sep  5 2022, 14:08:36) [MSC v.1933 64 bit (AMD64)] on win32
 - KiField version [e.g. 0.1.22]
 - 1.0.0
 - KiCad version
7.0.0
**Additional context**
Add any other context about the problem here.
This program is very useful! The structure of the library files has been changed for V7, it looks simpler

thestumbler, 2023-07-18:
Any update on this? I just spend a couple hours crafting a merged BOM spreadsheet into the KiField-generated exported spreadsheet, only to find I couldn't reimport it back again. Apparently because I'm using KiCad 7.  Any workarounds in the meantime?

devbisme, 2023-07-19:
I updated this program last year when KiCad 6 appeared with new file formats. Now they've updated file formats (or something) and broken things again. I guess it will remain broken until I run out of more interesting things to do.

thestumbler, 2023-07-19:
I did some digging around, and confirmed my suspicion that the schematic format was updated in v6. I'm not aware of changes in v7. Also, I'm not sure my error messages were exactly the same as @dilydaly1 reported. Let me re-run and see if the messages give any clues

thestumbler, 2023-07-19:
Similar, but not identical errors. Here is the error message running in Windows. Looks like @dilydaly1 was updating a library file, while I was trying to update a schematic file.
```
C:\files\projs\hybo\eda\robot-interface-mini>kifield -x fields.xlsx -i robot-interface-mini.kicad_sch -w
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\owner\.pyenv\pyenv-win\versions\3.11.4\Scripts\kifield.exe\__main__.py", line 7, in <module>
  File "C:\Users\owner\.pyenv\pyenv-win\versions\3.11.4\Lib\site-packages\kifield\__main__.py", line 149, in main
    kifield(
  File "C:\Users\owner\.pyenv\pyenv-win\versions\3.11.4\Lib\site-packages\kifield\kifield.py", line 1554, in kifield
    insert_part_fields(
  File "C:\Users\owner\.pyenv\pyenv-win\versions\3.11.4\Lib\site-packages\kifield\kifield.py", line 1518, in insert_part_fields
    insertion_function(
  File "C:\Users\owner\.pyenv\pyenv-win\versions\3.11.4\Lib\site-packages\kifield\kifield.py", line 1209, in insert_part_fields_into_sch_V6
    component.copy_field("Reference", field_name)
  File "C:\Users\owner\.pyenv\pyenv-win\versions\3.11.4\Lib\site-packages\kifield\sch.py", line 497, in copy_field
    id = find_by_key("id", dst_field["prop"])[0]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range

```
And here is the error when running in WSL / Ubuntu
```
rclott@stumbler robot-interface-mini$ kifield -x fields.xlsx -i robot-interface-mini.kicad_sch -w
Traceback (most recent call last):
  File "/home/rclott/.pyenv/versions/3.10.10/bin/kifield", line 33, in <module>
    sys.exit(load_entry_point('kifield==1.0.0', 'console_scripts', 'kifield')())
  File "/home/rclott/.pyenv/versions/3.10.10/lib/python3.10/site-packages/kifield/__main__.py", line 149, in main
    kifield(
  File "/home/rclott/.pyenv/versions/3.10.10/lib/python3.10/site-packages/kifield/kifield.py", line 1554, in kifield
    insert_part_fields(
  File "/home/rclott/.pyenv/versions/3.10.10/lib/python3.10/site-packages/kifield/kifield.py", line 1518, in insert_part_fields
    insertion_function(
  File "/home/rclott/.pyenv/versions/3.10.10/lib/python3.10/site-packages/kifield/kifield.py", line 1209, in insert_part_fields_into_sch_V6
    component.copy_field("Reference", field_name)
  File "/home/rclott/.pyenv/versions/3.10.10/lib/python3.10/site-packages/kifield/sch.py", line 497, in copy_field
    id = find_by_key("id", dst_field["prop"])[0]
IndexError: list index out of range
```

devbisme, 2023-08-11:
I think I have a release candidate that solves the KiCad7 problems. You can install it like this:
```bash
pip install git+https://github.com/devbisme/KiField
```
If you find errors, I'll try to fix them. If not, I'll prepare a release on PyPi.

closed 2023-09-10 by devbisme