devbisme/KiField #79: UnicodeDecodeError: 'gbk' codec can't decode byte 0xa8 in position 1144: illegal multibyte sequence

lyrain2009, 2023-12-01:
**Screenshots**
[image]
**Desktop (please complete the following information):**
 - OS [Win11]
 - Python version python3.8
 - KiField version [1.0.1]

lyrain2009, 2023-12-01:
Kicad info:
Kicad Application: KiCad x64 on x64
Version: 7.0.9, release build

devbisme, 2023-12-01:
Since you've identified the problem and have a potential fix, why not just issue a pull request?

lyrain2009, 2023-12-05:
> Since you've identified the problem and have a potential fix, why not just issue a pull request?
Thanks.

I find it's not a good ideal to modify source file,because all the open function in source shoud be added "encoding='utf-8' ".

the reason of UnicodeDecodeError is the system encoding method,because when python open file,it use system default encoding.
 I use windows in china,the default encoding is 'gbk',so when python open file ,it use 'gbk' codec to decode utf-8 file that cause error.
So the solution is set  Python to utf-8 mode,just set the PYTHONUTF8=1 environment variable,no need to modify source file.
[Python: Use the UTF-8 mode on Windows!](https://dev.to/methane/python-use-utf-8-mode-on-windows-212i)